riscv-software-src/riscv-isa-sim #376: How to use GDB in my case?

geonheec, 2020-01-08:
Hi, I want to use riscv64-unknown-linux-gnu-gdb in my computer.
(Ubuntu 16.04, kernel 4.15.0, x86_64 architecture)
I already built and installed "riscv-gnu-toolchain" and "riscv-tools" in _**riscv64-unknown-linux-gnu**_ mode. 

I also tried "Debugging With Gdb" in https://github.com/riscv/riscv-isa-sim and it worked well.
But I can't find a way to use gdb in that way for my legacy C program.

[1] Which compile flag is mandatory? My legacy Makefile uses **-Wall -O1 -static** flags.

[2] How can I write the .lds for my program??
When I used -T flag with the spike.lds in "Debugging with Gdb" example, compilation failed with following message.

> /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:75: undefined reference to `__preinit_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:75: undefined reference to `__preinit_array_end'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:75: undefined reference to `__preinit_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:75: undefined reference to `__preinit_array_end'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:77: undefined reference to `__preinit_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:86: undefined reference to `__init_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:86: undefined reference to `__init_array_end'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:86: undefined reference to `__init_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:86: undefined reference to `__init_array_end'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:87: undefined reference to `__init_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /opt/riscv/sysroot/usr/lib/libc.a(elf-init.o): in function `__libc_csu_fini':
/home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:98: undefined reference to `__fini_array_end'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:98: undefined reference to `__fini_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:98: undefined reference to `__fini_array_start'
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /home/ghcho/riscv-gnu-toolchain/riscv-glibc/csu/../csu/elf-init.c:98: undefined reference to `__fini_array_end'
collect2: error: ld returned 1 exit status

[3] How should I modify the bolded portion below?
$ spike --rbb-port=9824 **_-m0x10000000:0x20000_** myProgram

[4] Do I need to modify spike.cfg file in "Debugging With Gdb" example?

timsifive, 2020-01-08:
For single-hart instances of spike, the example config should be all you need.

All your other questions pertain to getting a program to run on spike. There is nothing special about a program that can be debugged. Whatever spike is executing, you can connect to gdb and debug it. The example in the README is intentionally tiny so that it fits.

You should focus on building a program that runs in spike.

geonheec, 2020-01-14:
Thanks for answer !
Have a nice day.

closed 2020-01-14 by geonheec