riscv-software-src/riscv-isa-sim #380: How to use gdb with my 32bit program?

geonheec, 2020-01-11:
I followed the guide from README in this repo.
(But I did not use spike.lds and just compile my program with riscv32-unknown-linux-gnu-gcc.)

The following commands are what I tried. (bomb is my riscv32 executable)
> spike --rbb-port=9824 ~/pk **bomb**
> openocd -f spike.cfg
> riscv32-unknown-linux-gnu-gdb **bomb**
> (gdb) target remote localhost:3000

but it prints
> bfd requires flen 8, but target has flen 0

I can't find what's the problem....

timsifive, 2020-01-13:
It looks like you're using an older gdb, which doesn't like talking to 32-bit targets unless you specify a debug file first. Solutions:
1. Specify your binary with gdb's `file` command before connecting to OpenOCD.
2. Use a recent gdb from git://sourceware.org/git/binutils-gdb.git

geonheec, 2020-01-14:
Thanks for answer!
I solve the problem.

closed 2020-01-14 by geonheec

intelligenttoasters, 2020-07-19:
For anyone else struggling with this error message (as I have for 2 days), note that regardless of the GCC options used to compile your code, the parameters used to build the toolchain may override the settings. When configuring the toolchain, use this for 4-byte (single precision) floats :
./configure --prefix=/opt/riscv --with-arch=rv32if --with-abi=ilp32f
or
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
for no hard floats, matching the target flen of zero.