riscv-software-src/riscv-isa-sim #382: viota.m yields unexpected result

gsauthof, 2020-01-13:
This Spike session shows an unexpected result for `viota.m`:

```
bbl loader
: 
core   0: 0x00000000000101b8 (0x003872d7) vsetvli t0, a6, e8,m8
: reg 0 t0
0x0000000000000010
: reg 0 a6
0x0000000000000010
: 
core   0: 0x00000000000101bc (0x5e00b857) vmv.v.i v16, 1
: vreg 0 16
VLEN=128 bits; ELEN=32 bits
v16 : [3]: 0x01010101  [2]: 0x01010101  [1]: 0x01010101  [0]: 0x01010101  
: 
core   0: 0x00000000000101c0 (0x53082457) viota.m v8, v16
: vreg 0 8
VLEN=128 bits; ELEN=32 bits
v8  : [3]: 0x02020202  [2]: 0x02020201  [1]: 0x01010101  [0]: 0x01010100
```

Since the least significant bit of each 8 bit sized element of v16 is set, I expect that `viota.m` fills v8 like this:

```
v8  : [3]: 0x0f0e0d0c [2]: 0x0b0a0908  [1]: 0x07060504  [0]: 0x03020100
```

The RISCV V specification 0.8 reads:

> The viota.m instruction reads a source vector mask register and writes to each element of the destination vector register group the sum of all the least-signicant bits of elements in the mask register whose index is less than the element, e.g., a parallel prefix sum of the mask values.

I'm testing this with yesterdays HEAD of the master branch (bb1cd8f) plus the fixes from #381.

aswaterman, 2020-01-13:
You have LMUL=8, so every bit of v16 matters, not every 8th bit.  I believe you'd see the result you expect if you use LMUL=1.

gsauthof, 2020-01-13:
Indeed, with `vsetvli t0, a6, e8` I get the expected result.

You mean that whith `viota.m vd, vs2, vm` the `vs2` operand is interpreted like `vm` and not as a register group. Thus the elements of vs2 aren't standard width, but `SEW/LMUL` sized - and only the first register of a register group is used.

Perhaps this could be made more explicit in the spec.

closed 2020-01-13 by gsauthof

aswaterman, 2020-01-13:
Yeah, that's what the spec means by describing vs2 as a "source vector mask register" rather than a "source vector register".

gsauthof, 2020-01-15:
Fair enough.

Although it wouldn't hurt to add a more formal definition to that `viota.m` Section, i.e. in addition to the existing prose.

For example a pseudo-code snippet such as:

    for (i=0; i < vl; ++i)
        if ((vm >> MLEN*i) & 1)
            if (i)
                vd[i] = vd[i-1] + ((vs2 >> MLEN*(i-1)) & 1);
            else
                vd[0] = 0;