riscv-software-src/riscv-isa-sim #384: Is it impossible to simulate rv32ia linux program with spike?

geonheec, 2020-01-19:
Hi, I'm trying to using spike to simulate a linux program.
So, I installed rv32ia arch **riscv-gnu-toolchain**.
Then I installed riscv-isa-sim by following instructions.
```
$ apt-get install device-tree-compiler
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV
$ make
$ [sudo] make install
```

Next, I installed riscv-pk by following instructions.
```
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV --host=riscv32-unknown-linux-gnu
$ make
$ make install
```

After the installation, I tried to simulate my linux program by `spike pk test` but the spike stopped without outputting anything.

So I tried `spike --isa=rv32ia pk test` and the following output is appeared.
```
bbl loader
z  00000000 ra 000102e4 sp 7fbe4da0 gp 00012800
tp 00000000 t0 00000000 t1 00000000 t2 00012270
s0 00000000 s1 00000000 a0 00010394 a1 00000001
a2 7fbe4da4 a3 000103c8 a4 00010458 a5 00000000
a6 7fbe4da0 a7 00000000 s2 00000000 s3 00000000
s4 00000000 s5 00000000 s6 00000000 s7 00000000
s8 00000000 s9 00000000 sA 00000000 sB 00000000
t3 ffffffff t4 00000000 t5 00000000 t6 00000000
pc 0001028c va fffffffe insn       ffffffff sr 80046020
Misaligned instruction access!
```

As well as my linux program, even the _hello world_ program does not run as above.

How can I simulate 32bit linux program?

aswaterman, 2020-01-19:
Did you compile the test program with the -static flag? pk doesn’t support dynamically linked executables.

geonheec, 2020-01-19:
Yes, I compiled the test program with the `-Wall -O1 -static flag -march=rv32ia`

closed 2020-01-26 by geonheec