riscv-software-src/riscv-isa-sim #389: BUG Illegal instruction when reading time CSR

victorspgl, 2020-01-31:
Hi,

while trying to boot linux in spike with the bbl, I have found that spike is not able to read time, mtime and stime CSR. Instead of returning a 0, as spike did in the past, now it throws an illegal instruction.

I left you here the instruction that launches the illegal exception
`0xc01027f3 csrr    a5, time`

aswaterman, 2020-01-31:
Privileged code is expected to trap and emulate this instruction by reading
the memory-mapped time CSR that lives in the uncore. This more closely
matches what real systems do.

pk and bbl provide this functionality, so user programs running on pk or
OSes running on bbl can read the time CSR.

victorspgl, 2020-02-01:
Ok, illegal instruction is a correct implementation if TM bit in mcounteren register is always clear. So, I'm closing the issue.

However, you can still implement a csrr to time as ISA explains: _"Implementations can convert reads of the time CSR into loads to the memory-mapped mtime register, or emulate this functionality in M-mode software"_

closed 2020-02-01 by victorspgl