riscv-software-src/riscv-isa-sim #401: Issue with pthread library (bad syscall #98)

federicorossifr, 2020-02-19:
I'm trying to execute a very little test program using pthread library:
```c++
#include <iostream>
#include <cstdlib>
#include <pthread.h>

using namespace std;

#define NUM_THREADS 5

void *PrintHello(void *threadid) {
   long tid;
   tid = (long)threadid;
   cout << "Hello World! Thread ID, " << tid << endl;
   pthread_exit(NULL);
}

int main () {
   pthread_t threads[NUM_THREADS];
   int rc;
   int i;
   
   for( i = 0; i < NUM_THREADS; i++ ) {
      cout << "main() : creating thread, " << i << endl;
      rc = pthread_create(&threads[i], NULL, PrintHello, (void *)i);
      
      if (rc) {
         cout << "Error:unable to create thread," << rc << endl;
         exit(-1);
      }
   }
   pthread_exit(NULL);
}
```
I am able to compile it using the toolchain from github. However, when I execute it on the spike simulator using:
`spike pk test`
I get the following error:
`bad syscall #98!`

If I comment out the line with the `pthread_create` invocation, everything is fine.

aswaterman, 2020-02-19:
pk does not support pthreads, so it’s failing on pthread_create.

colinschmidt, 2020-02-19:
This is a better issue for https://github.com/riscv/riscv-pk.
That being said the proxy kernel doesn't support pthreads, you can see a list of supported syscalls here: https://github.com/riscv/riscv-pk/blob/master/pk/syscall.h