riscv-software-src/riscv-isa-sim #402: Illegal AR option when make

rayjyh, 2020-02-20:
Make fails with the following error:

Building project riscv-isa-sim
+ /usr/local/bin/make
In file included from ../fesvr/tsi.cc:1:
../fesvr/tsi.h:40:10: warning: 'chunk_align' overrides a member function
      but is not marked 'override' [-Winconsistent-missing-override]
  size_t chunk_align() { return 4; }
         ^
../fesvr/htif.h:36:18: note: overridden virtual function is here
  virtual size_t chunk_align() = 0;
                 ^
In file included from ../fesvr/tsi.cc:1:
../fesvr/tsi.h:41:10: warning: 'chunk_max_size' overrides a member
      function but is not marked 'override'
      [-Winconsistent-missing-override]
  size_t chunk_max_size() { return 1024; }
         ^
../fesvr/htif.h:37:18: note: overridden virtual function is here
  virtual size_t chunk_max_size() = 0;
                 ^
2 warnings generated.
ar: illegal option combination for -r
usage:  ar -d [-TLsv] archive file ...
	ar -m [-TLsv] archive file ...
	ar -m [-abiTLsv] position archive file ...
	ar -p [-TLsv] archive [file ...]
	ar -q [-cTLsv] archive file ...
	ar -r [-cuTLsv] archive file ...
	ar -r [-abciuTLsv] position archive file ...
	ar -t [-TLsv] archive [file ...]
	ar -x [-ouTLsv] archive [file ...]
make: *** [Makefile:313: libfesvr.a] Error 1
make: *** Waiting for unfinished jobs....

aswaterman, 2020-02-21:
You need GNU `ar` to build these projects.

rayjyh, 2020-02-21:
@aswaterman I did have `ar` installed though. Otherwise it wouldn't have given me the usage options. I reinstalled binutils, the build still failed.

aswaterman, 2020-02-21:
Emphasis on GNU `ar`, as opposed to some other vendor's `ar`:

```$ ar --version
GNU ar (GNU Binutils for Ubuntu) 2.26.1
Copyright (C) 2015 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
```

aswaterman, 2020-02-21:
If you figure out how to make it work with your vendor's `ar`, we'll gladly accept a PR (provided it continues to work with GNU `ar`).

rayjyh, 2020-03-02:
Thanks @aswaterman. I have gnu `ar` installed now. But I got the following error:

> Building project riscv-isa-sim
+ /usr/local/bin/make
In file included from ../fesvr/tsi.cc:1:
../fesvr/tsi.h:40:10: warning: 'chunk_align' overrides a member function but is
      not marked 'override' [-Winconsistent-missing-override]
  size_t chunk_align() { return 4; }
         ^
../fesvr/htif.h:36:18: note: overridden virtual function is here
  virtual size_t chunk_align() = 0;
                 ^
In file included from ../fesvr/tsi.cc:1:
../fesvr/tsi.h:41:10: warning: 'chunk_max_size' overrides a member function but
      is not marked 'override' [-Winconsistent-missing-override]
  size_t chunk_max_size() { return 1024; }
         ^
../fesvr/htif.h:37:18: note: overridden virtual function is here
  virtual size_t chunk_max_size() = 0;
                 ^
2 warnings generated.
ld: warning: ignoring file ./libfesvr.a, building for macOS-x86_64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture x86_64:
  "load_elf(char const*, memif_t*, unsigned long long*)", referenced from:
      _main in elf2hex.o
  "htif_hexwriter_t::htif_hexwriter_t(unsigned long, unsigned long, unsigned long)", referenced from:
      _main in elf2hex.o
  "vtable for htif_hexwriter_t", referenced from:
      _main in elf2hex.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "vtable for memif_t", referenced from:
      _main in elf2hex.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "operator<<(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, htif_hexwriter_t const&)", referenced from:
      _main in elf2hex.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:313: elf2hex] Error 1

aswaterman, 2020-03-02:
Not sure; it works fine for me on High Sierra.  If you figure it out, please feel free to recommend a patch.

zarubaf, 2020-11-05:
I have had a similar experience on Catalina. Problem was that `clang/ld` wasn't recognizing the ELF's produced with Homebrew's (GNU) `binutils`/`ar`. Using the default system `ar` solved it for me, therefore I've uninstalled Homebrew's binutils:

```
brew uninstall binutils
```

Which left my system's `ar`: 
```
❯ which ar
/usr/bin/ar
```

Maybe that helps.