riscv-software-src/riscv-isa-sim #408: satp don't support ASID

weiqi7777, 2020-03-04:
hi:
  i look the new riscv privileged spec.
  for satp, add ASID field.
  but I look for the spike code, the satp don't support asid, and writing asid field value is ignored.

aswaterman, 2020-03-04:
Yes. This is a legal implementation, since ASIDs are optional. A future
version of Spike might support ASIDs optionally, but really all that means
is the field would be writable. There would be no other functional effect,
since Spike does not model the same kind of TLB you would build in a
practical hardware implementation.

scottj97, 2020-03-29:
Andrew, Spike has some command-line options to configure things like --isa, --priv, and --varch, but these feel haphazard and inconsistent. Is there a general strategy around configuring Spike to match a platform spec? Would it make sense to use the riscv-config effort to achieve this? Or is that just not a goal of Spike?

I ask here because ASID width is one such platform parameter.

aswaterman, 2020-03-29:
I agree with your characterization.  Not sure if the config approach will work out as an input format to configure Spike, but it's something we can explore.  In the mean time, it's probably going to continue to be a hodgepodge.

aswaterman, 2020-03-29:
To answer the other part of your question, emulating all conceivable RISC-V implementations is a non-goal.  This is to say that Spike does have a de facto platform (or space of platforms).  We should write that down.