riscv-software-src/riscv-isa-sim #411: Question about Installation

PeppaCat, 2020-03-05:
Hi, I suppose I met  some problem when installing spike, could you please give me some advice?
Environment: ubuntu18.04
Here is the step I take to first install [**RISC-V Open Source GNU Toolchain** ](https://mindchasers.com/dev/rv-getting-started)then install **riscv-isa-sim**.
```
$ cd /build
$ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
$ cd /build/riscv-gnu-toolchain/
$ mkdir build; cd build
$ ../configure --prefix=/opt/riscv32 --with-arch=rv32im --with-abi=ilp32
```
Here is the output to check what has been done :
```
$ riscv32-unknown-elf-gcc -dumpmachine
riscv32-unknown-elf
$ riscv32-unknown-elf-gcc -print-sysroot
/opt/riscv32/riscv32-unknown-elf
$ riscv32-unknown-elf-gcc -print-libgcc-file-name
/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/libgcc.a
$ riscv32-unknown-elf-gcc -print-search-dirs
/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/
programs: =/opt/riscv32/libexec/gcc/riscv32-unknown-elf/9.2.0/:/opt/riscv32/libexec/gcc/riscv32-unknown-elf/9.2.0/:/opt/riscv32/libexec/gcc/riscv32-unknown-elf/:/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/:/opt/riscv32/lib/gcc/riscv32-unknown-elf/:/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/riscv32-unknown-elf/9.2.0/:/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/
libraries: =/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/:/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/lib/riscv32-unknown-elf/9.2.0/:/opt/riscv32/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/lib/:/opt/riscv32/riscv32-unknown-elf/lib/riscv32-unknown-elf/9.2.0/:/opt/riscv32/riscv32-unknown-elf/lib/:/opt/riscv32/riscv32-unknown-elf/usr/lib/riscv32-unknown-elf/9.2.0/:/opt/riscv32/riscv32-unknown-elf/usr/lib/
```
Then I try to install riscv-isa-sim.
As mensioned in the github page, “ we should set RISCV environment variable is set to the RISC-V tools install path“.
So in the **riscv-gnu-toolchain/build** folder:
```
$export PATH=/opt/riscv32/bin:$PATH
$ apt-get install device-tree-compiler
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV
$make
nothing to be done for 'all'.
```
So I find I failed to install spike, and now I got no idea how to fix it.

aswaterman, 2020-03-05:
```
$ make
nothing to be done for 'all'.
```

suggests to me that you successfully built Spike already.  What happens when you run `make install`?

PeppaCat, 2020-03-05:
It suggests:make: Nothing to be done for 'install'.(I think this means that it failed to build?)
Still, when I type spike, it suggests that it is not installed.