riscv-software-src/riscv-isa-sim #413: Issue with vector loads

isVicjiar, 2020-03-05:
When trying to run a simple memory code that loads from a region, I do a vsetvli to set both VL to 10 and sew to 16, but when seeing the instruction results, this data appears in the vreg:

`...0000000011111111 0000000011111111 0000000000000000 0000000000000000`

It seems like it's using 32 as sew, as the correct result would be `1111 1111 1111 1111 0000 0000 0000 0000`.

I tried the same test using vsetvli to set VL to 10 and sew to 32, and the results were:

`...0000000011111111 0000000011111111 0000000000000000 0000000000000000`

And then same VL and 8 as sew:

`...0000000000000000 0000000000000000 0000000000000000 0000000000000000`

So it seems that it always writes 32 bits of the register (as if it was s32 instead of s64). Varch is "v16384:e64:s64".

isVicjiar, 2020-03-05:
I checked and it happens with every instruction.

It would be great if you could help me :smile: 

Thanks!

aswaterman, 2020-03-05:
@chihminchao

chihminchao, 2020-03-10:
@isVicjiar 
could you provide an snip of problematic code or an  test is program is better.

isVicjiar, 2020-03-10:
Sure! Thanks!
[mem_test2_16.txt](https://github.com/riscv/riscv-isa-sim/files/4311383/mem_test2_16.txt)

chihminchao, 2020-03-13:
@isVicjiar 
I am curious about the arch parameter ""v16384:e64:s64". Current implementation in spike should support vlen <= 4096. 
By the way,  at present, vlen != slen is not fully supported.  It may works for load/store but should failed for the other instructions

Marc43, 2020-03-26:
Hi @chihminchao , if VLEN is expressed in bits this is not enough for us, our vector registers are bigger, do you know any workaround for this issue? Thanks!!

aswaterman, 2020-03-26:
@chihminchao if VLEN != SLEN is not fully supported, we should issue an error immediately, rather than silently doing the wrong thing.

@Marc43 I agree Spike should be able to support very large VLEN.  However, large VLEN usually also means VLEN != SLEN, so it might take some time to fully support the kind of machine you want to model.

opalomar, 2020-03-26:
But, can the VLEN limit be removed for VLEN==SLEN? That is what we are actually implementing.

Thanks

aswaterman, 2020-03-26:
@opalomar I think it should be straightforward to remove the VLEN limit when VLEN==SLEN, but I defer to @chihminchao on the details.

opalomar, 2020-03-26:
@aswaterman Excellent, thanks a lot.

chihminchao, 2020-03-27:
@opalomar 
    Yes, the max vlen can be relaxed to spec definition. The original limitation is used to reduce testing time and 4096 seems to be a reasonable scale for most implementation.
@aswaterman
    I will send vlen != slen checking patch later.

qq332982511, 2022-02-19:
is it done?

aswaterman, 2022-02-19:
Two years later?  Let's just say yes until proven otherwise.

closed 2022-02-19 by aswaterman