riscv-software-src/riscv-isa-sim #416: riscv-isa-sim/debug_rom.h what is the difference between this and debug_rom.bin？

Toxic-Scofield, 2020-03-09:
Hi！
     As we all know ，debug_rom.bin is machine code that cpu can understand. It is the as same as real instruction in Icache. The debug_rom.bin can correspond to debug_rom.dump. But I don't know what debug_rom.h is. They are similar but different. It seems that rocketchip/DebugRomContents just puts it in.  Dose anybody know about that and how rocketchip uses it?

Here is rocketchip/DebugRomContents.
`// This file was auto-generated by 'make publish' in debug/ directory.
package freechips.rocketchip.devices.debug
object DebugRomContents {
  def apply() : Array[Byte] = { Array (
  0x6f, 0x00, 0xc0, 0x00, 0x6f, 0x00, 0x80, 0x03, 0x6f, 0x00, 0x40, 0x04,
  0x0f, 0x00, 0xf0, 0x0f, 0x73, 0x10, 0x24, 0x7b, 0x73, 0x24, 0x40, 0xf1,
  0x23, 0x20, 0x80, 0x10, 0x03, 0x44, 0x04, 0x40, 0x13, 0x74, 0x34, 0x00,
  0xe3, 0x08, 0x04, 0xfe, 0x13, 0x74, 0x14, 0x00, 0x63, 0x08, 0x04, 0x00,
  0x73, 0x24, 0x20, 0x7b, 0x23, 0x22, 0x00, 0x10, 0x67, 0x00, 0x00, 0x30,
  0x73, 0x24, 0x40, 0xf1, 0x23, 0x24, 0x80, 0x10, 0x73, 0x24, 0x20, 0x7b,
  0x73, 0x00, 0x20, 0x7b, 0x23, 0x26, 0x00, 0x10, 0x73, 0x00, 0x10, 0x00
  ).map(_.toByte) }
}`
Here is debugrom.bin(format has been modified, but contents are the same as .bin)
`// Auto-generated code
const int reset_vec_size = 38;
uint32_t reset_vec[reset_vec_size] = {
    0x00c0006f,
    0x07c0006f,
    0x04c0006f,
    0x0ff0000f,
    0x7b241073,
    0x7b351073,
    0x00000517,
    0x00c55513,
    0x00c51513,
    0xf1402473,
    0x10852023,
    0x00a40433,
    0x40044403,
    0x00147413,
    0x02041c63,
    0xf1402473,
    0x00a40433,
    0x40044403,
    0x00247413,
    0xfa041ce3,
    0xfd5ff06f,
    0x7b351073,
    0x00000517,
    0x00c55513,
    0x00c51513,
    0x10052623,
    0x7b302573,
    0x00100073,
    0x10052223,
    0x7b202473,
    0x7b302573,
    0xa85ff06f,
    0xf1402473,
    0x10852423,
    0x7b202473,
    0x7b302573,
    0x7b200073,
    0x00000000
};`

Thanks!

aswaterman, 2020-03-09:
The rocket-chip version can avoid the FENCE.I because of some microarchitecture-specific knowledge.  It's also not kept up to date with the Spike one, so there may be other differences.

Toxic-Scofield, 2020-03-09:
> The rocket-chip version can avoid the FENCE.I because of some microarchitecture-specific knowledge.  It's also not kept up to date with the Spike one, so there may be other differences.

Thanks

closed 2020-03-12 by Toxic-Scofield