riscv-software-src/riscv-isa-sim #418: Some instructions are not decoded as illegal instruction when running in rv32imc mode

Evgenii666, 2020-03-10:
I am running an illegal instruction test and expect that some instructions to be decoded as illegal instruction. However, spike decoded is as a vector instruction. This issue is similar to https://github.com/riscv/riscv-isa-sim/issues/313 and https://github.com/riscv/riscv-isa-sim/issues/325.

core   0: 0xffffffff80003a22 (0x0953f287) vlse.v  v5, (t2), s5, v0.t
core   0: 0xffffffff8000c646 (0x6ceaf207) vlxseg4e.v v4, (s5), v14, v0.t

aswaterman, 2020-03-10:
@chihminchao

chihminchao, 2020-03-13:
@Evgenii666 
Do you use the HEAD version  ?

Evgenii666, 2020-03-13:
At the time of writing the issues, I was using the latest version.

chihminchao, 2020-03-17:
@Evgenii666 
I can't reproduce it on my site with --isa=rv32imc and it trigger illegal exception.