riscv-software-src/riscv-isa-sim #422: SPIKE reports instruction disassembly incorrectly

KrishnaRadhakrishnan, 2020-03-13:
Spike prints the instruction csrrw as a store instruction in the disassembly. It reports memory access information corresponding to a csrrw instuction, which is unexpected. Please note that we observe this behaviour only for store instructions. Load instruction is unaffected.

core   0: 0x00000000400012c0 (0x00a50023) sb      a0, 0(a0)
core   0: exception trap_store_access_fault, epc 0x00000000400012c0
core   0:           tval 0x0000000040004008
core   0: 0x00000000400000f0 (0x34011173) csrrw   sp, mscratch, sp
3 0x00000000400000f0 (0x34011173) x 2 0x0000000040007a00 **mem 0x0000000040004008 0x8**

The test triggers a 'Store access fault' exception which starts the trap handler execution
Please refer the below PMP test sequence:

	400012c0:	00a50023          	sb	a0,0(a0) 
00000000400000f0 <trap_entry>:
	400000f0:	34011173          	csrrw	sp,mscratch,sp

Please refer the spike log attached.

We have performed tests generating 'illegal instruction' exception, store/load instruction generating 'Address misaligned' exceptions and SPIKE was reporting the same csrrw instruction correctly in such cases.

aswaterman, 2020-03-13:
@chihminchao do you think this is a problem with the commit-log code? (I didn’t have time to internalize the details.)

chihminchao, 2020-03-17:
Yes. It is commitlog issue and I will send PR later.

chihminchao, 2020-03-19:
It has been fix in https://github.com/riscv/riscv-isa-sim/pull/425.