riscv-software-src/riscv-isa-sim #424: Issue with vsetvli happening again

dimas-dimas, 2020-03-16:
I am trying to run some tests using the V extension using the latest version of spike and  I have run into the same problems reported in issue #381, vsetvli yields a trap_illegal_instruction. In fact, if I go back to commit d15d781, which is the one related to #381, everything works as expected.

chihminchao, 2020-03-16:
Could you check the vector unit initialization code ? The position of mstatus.vs bit has been changed in 7221b28

dimas-dimas, 2020-03-16:
Could you be more specific about what you would like me to check? I do not have much experience with spike.

chihminchao, 2020-03-17:
After d15d781 , there is an  2bit field, vs, in  mstatus to enable vector unit.  You should initialize it correctly before executing any vector instructions.    However, the field position is changed at 7221b28.  The old is mstatus[24:23]  but the new is mstatus[10:9].  The initialization code must be modified to match this change.