riscv-software-src/riscv-isa-sim #427: Hard-coded path to DTC breaks packaging

joelsherrill, 2020-03-21:
The RTEMS Project includes Spike in our package building system as part of a RISC-V toolset. The packaging tool (as is common) builds and installs into a temporary location which has a prefix before the ultimate intended prefix (e.g. /tmp/building/${prefix}). We package DTC with Spike so this results in the configure.ac probe configuring dtc to use a temporary directory, not the real install location.

IMO, hard coding a path like this makes spike sensitive to post build issues. If a user installs dtc in a directory like /usr/local/dtc-1.2 and /usr/local/dtc-1.4, spike is dependent on this and spike must be rebuilt to use a newer dtc.

I would suggest and am happy to implement that dtc be invoked as a simple program name without full path. The alternative is that there be a configure option to use the simple program name instead of the full path.

Please make a suggestion as to your preferred solution or implement it yourself. It appears to be fairly trivial to just set DTC to dtc in configure.ac and solve the problem from my perspective.

Thanks.

aswaterman, 2020-03-21:
Sounds reasonable to me.  Would you please make the PR?

joelsherrill, 2020-03-22:
Pull request created. Thanks for the quick feedback.

aswaterman, 2020-03-22:
Thank you. Merged, so closing this issue.

closed 2020-03-22 by aswaterman