riscv-software-src/riscv-isa-sim #431: Different behavior of interactive debug mode and normal mode.

JaewonHur, 2020-03-23:
When the spike executes misaligned lr.d followed by sc.w at the same address, interactive debug mode and normal mode show different behavior.

In normal mode, sc.w fails as expected, however it succeeds in interactive debug mode.
As a result, normal mode saves 1 to destination register but interactive mode saves 0.

I used Spike git version: 157143b5038ef2b92b1c70d445b566d57b6f5096.

All my stuff is as follow.
[test.zip](https://github.com/riscv/riscv-isa-sim/files/4368332/test.zip)

aswaterman, 2020-03-23:
Nice catch.  The implementations of LR.W and LR.D should not have acquired the load reservation before determining that the instruction might raise an exception.

khannaudit, 2020-06-08:
@aswaterman I think your fix for this issue may have been incomplete. 

Here, I'm doing an `LR.W` to an aligned address followed by an `SC.W` to a misaligned address and getting a page fault.

> core   3: 0xffffffc511a0a0ee (0x1002afaf) lr.w    t6, (t0)
> 0 0xffffffc511a0a0ee (0x1002afaf) x31 0x00000000184bd04b mem 0xffffffc500009000
> core   3: 0xffffffc511a0a0f2 (0xffd28293) addi    t0, t0, -3
> 0 0xffffffc511a0a0f2 (0xffd28293) x 5 0xffffffc500008ffd
> core   3: 0xffffffc511a0a0f6 (0x1992a32f) sc.w    t1, s9, (t0)
> core   3: exception trap_store_page_fault, epc 0xffffffc511a0a0f6
> core   3:           tval 0xffffffc500008ffd

Your fix https://github.com/riscv/riscv-isa-sim/commit/b4add16f4923a782f79b478d980791ec1d0bee06 addresses the alignment exception priority for LR but not SC

I believe the logic here needs to change but I'm not confident enough to propose a fix
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/sc_w.h#L3

aswaterman, 2020-06-08:
Agreed.

commit b4add16f (2020-03-23): Don't acquire load reservation in the event of a fault

I think this bug wasn't caught because OS code never steps over faulting
LR instructions in practice.  The exception is either fatal (in which case
the point is moot) or the LR is re-executed (in which case the point is
also moot).

Resolves #431
commit 090a083f (2020-06-08): Fix priority of misaligned exceptions for store-conditional

Previously, we unintentionally prioritized access faults and page faults.

Resolves #431
closed 2020-06-08 by aswaterman