riscv-software-src/riscv-isa-sim #435: fesvr: Assumes dmactive=1 write is immediately effective

mwachs5, 2020-03-29:
FESVR currently writes dmactive=1 then immediately begins other debug module accesses. The RISC-V Debug spec says:

"        <field name="dmactive" bits="0" access="R/W" reset="0">
            This bit serves as a reset signal for the Debug Module itself.

            0: The module's state, including authentication mechanism,
            takes its reset values (the \FdmDmcontrolDmactive bit is the only bit which can
            be written to something other than its reset value). Any accesses
            to the module may fail. Specifically, \FdmDmstatusVersion may not return
            correct data.

            1: The module functions normally. After writing 1, the debugger should
            poll \RdmDmcontrol until \FdmDmcontrolDmactive is high. Hardware may
            take an arbitrarily long time to initialize and will indicate completion
            by setting dmactive to 1.
"

FESVR needs to poll DMCONTROL to see dmactive =1 before proceeding

aswaterman, 2020-03-29:
Since it looks like we're intentionally leaving `dmactive` set for the entire execution (because when using the DTM HTIF, you don't care too much about power anyway), I think we only need to perform this polling once, in `producer_thread`, when we first set `dmactive`.  The other places that `dmactive` is set are red herrings, since it's already set.  Can you sanity-check my claim?

aswaterman, 2020-03-29:
I made a PR under the assumption that my previous comment is correct.  LMK.

commit 9e29fb64 (2020-03-29): When enabling the debug module, poll til it's really enabled

Resolves #435
commit acd953af (2020-03-30): When enabling the debug module, poll til it's really enabled

Resolves #435
PR #436 (2020-03-30): When enabling the debug module, poll til it's really enabled
Resolves #435
closed 2020-03-30 by aswaterman