riscv-software-src/riscv-isa-sim #440: Extracting memory of spike

FishCoding, 2020-04-03:
Hi, I am trying to extract a memory region doing a step by step simulation with a modified version of your repository. How would be the best way to extract this region if I am able during the execution to access to all the classes? 

Thanks in advance 😄

aswaterman, 2020-04-04:
All the memories (though there's only one memory in the default case) are available from the `mems` array in `main`.  The array is also passed into the `sim_t` constructor.

FishCoding, 2020-04-06:
This is what I am doing right now, it works without the pk. I also have a test with the proxy kernel where when doing the mem.read(....) it throws "trap_load_access_fault". The test with the proxy kernel without getting the memory finishes well. Do you have  any idea why this is happening?

closed 2020-06-10 by FishCoding