riscv-software-src/riscv-isa-sim #441: spike said vfwcvt.f.x.v is an illegal instruction?

xicongye, 2020-04-07:
Hi,
I want to run some tests on the latest version of spike & pk, here is my test code:
```
$ cat main.c
#include <stdio.h>
int16_t pSrc[4] = {1, 2, 3, 4};  
float pDst[4] = {0.0};  

int main()
{
  asm volatile(
    "li a0, 4\n"
    "vsetvli t0, a0, e16\n"
    "vle.v v0, (%0)\n"
    "vfwcvt.f.x.v v8, v0\n"                                                  
    :   
    : "r" (pSrc)
  );  

  return 0;
}
```
build & run:
```
$ riscv64-unknown-elf-gcc -march=rv64gcv -mabi=lp64d main.c
$ spike --isa=rv64imafdcv --varch=vlen:128,elen:64,slen:128  pk  a.out
bbl loader
z  0000000000000000 ra 0000000000010100 sp 000000007f7e9b40 gp 0000000000011c70
tp 0000000000000000 t0 0000000000000004 t1 000000000000000f t2 0000000000000000
s0 000000007f7e9b50 s1 0000000000000000 a0 0000000000000004 a1 000000007f7e9b58
a2 0000000000000000 a3 0000000000000010 a4 0000000000000001 a5 0000000000011bc8
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 000000000001016a va 000000008a059457 insn       8a059457 sr 8000000200046620
An illegal instruction was executed!
```
I checked the disassembly file.
```
$ riscv64-unknown-elf-objdump -D a.out > main.asm
$ cat main.asm
0000000000010156 <main>:
   10156:       1141                    addi    sp,sp,-16
   10158:       e422                    sd      s0,8(sp)
   1015a:       0800                    addi    s0,sp,16
   1015c:       f5818793                addi    a5,gp,-168 # 11bc8 <pSrc>
   10160:       4511                    li      a0,4
   10162:       004572d7                vsetvli t0,a0,e16,m1,d1
   10166:       0207f007                vle.v   v0,(a5)
   1016a:       8a059457                vfwcvt.f.x.v    v8,v0  
   1016e:       4781                    li      a5,0
   10170:       853e                    mv      a0,a5
   10172:       6422                    ld      s0,8(sp)
   10174:       0141                    addi    sp,sp,16
   10176:       8082                    ret
```
It looks like vfwcvt.f.x.v is an illegal instruction.
According to the riscv-v-spec(https://github.com/riscv/riscv-v-spec) , chapter 11.2, there are three situations that will cause an illegal instruction exception:
Situation 1:  my LMUL is set to 1, so 2*LMUL=2 is supported
> For all widening instructions, the destination element width must be a supported element width and the destination LMUL value must also be a supported LMUL value (≤8, i.e., current LMUL must be ≤4), otherwise an illegal instruction exception is raised.

Situation 2: vector registers I used in the test code is valid
> The destination vector register group must be specified using a vector register number that is valid for the destination’s LMUL value, otherwise an illegal instruction exception is raised.

Situation 3:  destination register I use in the test code didn't overlap the source register
> The destination vector register group cannot overlap a source vector register group of a different element width (including the mask register if masked), otherwise an illegal instruction exception is raised.

Do I miss something?
Thanks.

aswaterman, 2020-04-07:
cc @chihminchao

chihminchao, 2020-04-08:
@rickoco 
It is because you use e16, meaning float16.  The type is not supported in current spike implementation.  Thought the spec says that it should support all IEEE-754 type but there is no corresponding fp16 spec in scalar part.  I decide to wait for more completed spec for this part.

xicongye, 2020-04-08:
@chihminchao 
Thanks for your reply. According to the riscv-v-spec(https://github.com/riscv/riscv-v-spec) , chapter 14.16, `vfwcvt.f.x.v` instruction can convert signed integer to double-width float.
```
vfwcvt.f.x.v vd, vs2, vm # Convert signed integer to double-width float.
```
So, according to my understanding(correct me if I'm wrong), `vfwcvt.f.x.v` can convert 16-bit signed integer to 32-bit single-precise floating-point if the SEW is set to e16.

chihminchao, 2020-04-08:
@xicongye 
Yes, the source is integer but the instruction belongs to vector floating category. Current implementation checks element size and the existence of corresponding scalar floating extension in front of each vector floating instruction.  It means  we check something like
   element_ size == e16 &&  require_ext("float16") 

There are only two floating scalar extensions, 'F' and 'D. and the only two valid element size for vector floating instructions are  e32 and e64.

xicongye, 2020-04-08:
Got it. Thanks.

closed 2020-04-08 by xicongye