riscv-software-src/riscv-isa-sim #447: when core is in the privilege which is less than "M",then "mie" and "sie" are useless and "global interrupt switch" is always open

lizhirui, 2020-04-16:
when core is in the privilege which is less than "M",then "mie" and "sie" are useless and "global interrupt switch" is always open
the error is issued because of the line 408 and the line 412 of function "processor_t::take_interrupt" of file "processor.cc" in "riscv" directory.

reg_t m_enabled = state.prv < PRV_M || (state.prv == PRV_M && mie);
//if current privilege is less than M,m_enabled is always 1!
reg_t s_enabled = state.prv < PRV_S || (state.prv == PRV_S && sie);
//if current privilege is less than S,m_enabled is always 1!

I found that error in K210 chip which used about 2016 year version rocketchip core,so I guess HDL Code has the same problem.

aswaterman, 2020-04-16:
Yeah, that’s the spec...

lizhirui, 2020-04-16:
in the supervisor mode,I can't use sie to control the global interrupt switch is on or off,and althrough I set mie is 1 in M-Mode,in S-Mode,the interrupt switch is also always on.

aswaterman, 2020-04-16:
I can’t speak to what the K210 does, but the design is that sstatus.sie
masks all interrupts when you’re in S-mode, and it has no effect otherwise.

lizhirui, 2020-04-16:
> I can’t speak to what the K210 does, but the design is that sstatus.sie masks all interrupts when you’re in S-mode, and it has no effect otherwise.
> […](#)

actually,according to that code in spike,when state.prv = PRV_S,m_enabled = 1,and then
reg_t enabled_interrupts = pending_interrupts & ~state.mideleg & -m_enabled;
-m_enabled = 0xFFFFFFFFFFFFFFUL
if a interrupt is triggered and flagged in the "pending_interrupts",and if the corresponding bit in "state.mideleg" is 0,the "enabled_interrupts" variable of that bit must be 1,
and in line 176,enabled_interrupts |= pending_interrupts & state.mideleg & -s_enabled;
that's "|=".
So in S-Mode,even if mstatus.mie and sstatus.sie are all 0,the interrupt which isn't delegated to S-Mode and is triggered must make core entry the trap routine.

lizhirui, 2020-04-16:
Yeah,I found that spec "Interrupts for lower privilege modes are always disabled, whereas interrupts for higher privilege
modes are always enabled. Higher-privilege-level code can use separate per-interrupt enable bits
to disable selected interrupts before ceding control to a lower privilege level".
I didn't find that before.