riscv-software-src/riscv-isa-sim #450: Is there specified behavior for sc.d if there is an exception between lr.d and sc.d?

JaewonHur, 2020-04-21:
Hi, is there a defined behavior for sc.d if there is an exception between lr.d and sc.d?
(both operation accesses same address)

Again I think there is still mismatch between normal mode and interactive mode lrsc operation.

I used commit: d1b6eb11137d5fd23156bb32fe42eaec30b558c5
and here is the test code.

[test.zip](https://github.com/riscv/riscv-isa-sim/files/4509525/test.zip)

aswaterman, 2020-04-21:
The spec is that the sc.d _might_ succeed. sc.d is only guaranteed to
eventually succeed if the sequence is short enough, no exceptions occur,
etc. So both behaviors sound OK.

Spike will yield the load reservation when it breaks out of the simulation
loop in processor_t::execute. So a difference between interactive and
normal mode is not surprising to me. You could look into it further. If
making the behavior match isn’t too onerous, we should do so.

JaewonHur, 2020-04-22:
Thanks for reply.
let me report if I find solution.

JaewonHur, 2020-04-23:
Spike succeeds when divu insn is in LR/SC loop (_unconstrained_)
Boom fails.

[test.zip](https://github.com/riscv/riscv-isa-sim/files/4520585/test.zip)