riscv-software-src/riscv-isa-sim #451: Possible bug when executing vwmaccus_vx

Marc43, 2020-04-22:
HI,

I have found a potential error when executing vwmaccus_vx, macro `VI_WIDE_OP_AND_ASSIGN_MIX`, macro is called with the following arguments `VI_WIDE_OP_AND_ASSIGN_MIX(vs2, rs1, vd_w, *, +, int, int, uint)`.

Values of vs2 and rs1 are respectively: `0x7fc00000` and `0xd59f57ac`.
When vd_w is loaded from the register file the value is: `0x3fe000003fe00`

The code that calculates the value of an element is:

```
  default: { \
    sign_d##64_t vd_w = P.VU.elt<sign_d##64_t>(rd_num, i); \
    P.VU.elt<uint64_t>(rd_num, i, true) = \
      op1((sign_1##64_t)(sign_1##32_t)var0 op0 (sign_2##64_t)(sign_2##32_t)var1) + var2; \
    } \
```

The expansion of ` op1((sign_1##64_t)(sign_1##32_t)var0 op0 (sign_2##64_t)(sign_2##32_t)var1) + var2` would be `+((int64_t)(int32_t)vs2 * (uint64_t)(uint32_t)rs1) + vd_w`.

I have put a breakpoint there and see the value of the expression itself and the assigned value to the register file:

```
(gdb) p/x +((int64_t)(int32_t)vs2 * (uint64_t)(uint32_t)rs1) + vd_w
$18 = 0x6a9e42001503fe00
(gdb) p/x ((int64_t*)(p.VU.reg_file))[0]
$19 = 0xeade42001503fe00
(gdb) p/t +((int64_t)(int32_t)vs2 * (uint64_t)(uint32_t)rs1) + vd_w
$20 = 110101010011110010000100000000000010101000000111111111000000000
(gdb) p/t ((int64_t*)(p.VU.reg_file))[0]
$21 = 1110101011011110010000100000000000010101000000111111111000000000
```
Somehow the values differ, is this a expected behaviour? Do you know why is this happening?

Thanks,
Marc

aswaterman, 2020-04-22:
@chihminchao

Marc43, 2020-04-23:
Don't really know how this is not happening anymore. Thanks for your effort.

closed 2020-04-23 by Marc43