riscv-software-src/riscv-isa-sim #453: "spike --isa=RV32IMAFDC pk hello" can't output hello world !

rockcarry, 2020-04-24:
I buillt a toolchain for RV32IMC from riscv-gnu-toolchain project
./configure --prefix=/opt/riscv32 --with-arch=rv32imc --with-abi=ilp32

wrote a hello.c
#include <stdlib.h>
#include <stdio.h>

int main(void)
{
    printf("hello world !\n");
    return 0;
}

using riscv32-unknown-elf-gcc toolchain compile hello.c
riscv32-unknown-elf-gcc -Wall hello.c -o hello

run it:
spike --isa=RV32IMAFDC pk hello

there is no output.

if using riscv64-unknown-elf-gcc to compile it and run, it works fine:
riscv64-unknown-elf-gcc hello.c
spike --isa=RV64IMAFDC pk a.out
bbl loader
hello world !

spike does not support RV32 ?

hyf6661669, 2020-05-05:
Spike does support rv32, because I successfully use it to run elf file compiled by rv32 compiler.

hyf6661669, 2020-05-05:
I think maybe your riscv-proxy-kernel is not set to 32-bit, 
```
We assume that the RISCV environment variable is set to the RISC-V tools install path, and that the riscv-gnu-toolchain package is installed. Please note that building the binaries directly inside the source directory is not supported; you need to use a separate build directory.

$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV --host=riscv64-unknown-elf
$ make
$ make install
Alternatively, the GNU/Linux toolchain may be used to build this package, by setting --host=riscv64-unknown-linux-gnu.

By default, 64-bit (RV64) versions of pk and bbl are built. To built 32-bit (RV32) versions, supply a --with-arch=rv32i flag to the configure command.

The install step installs 64-bit build products into a directory matching your host (e.g. $RISCV/riscv64-unknown-elf). 32-bit versions are installed into a directory matching a 32-bit version of your host (e.g. $RISCV/riscv32-unknown-elf).
```

sahooora, 2021-12-13:
The same issu!
Did you find a way to solve it?

rockcarry, 2021-12-15:
> 

using 64bit is OK. the way to solove it, is using riscv 64bit. another is I written a riscv VM to run my code.

jundijiujieke, 2023-05-04:
I use the riscv64-unknown-elf-gcc to compile the hello.c
But i can't find pk version.
[image]

And when i run "spike pk hello", its result is strange.
[image]
Does anyone know how to solve this problem?

riscv1111, 2023-10-25:
> I compiled hello.c with riscv64-unknown-elf-gcc but I can't find the pk version. [image]
> 
> When I run "spike pk hello", it turns out strangely. [image]Does anyone know how to fix this?

How should this problem be solved？