riscv-software-src/riscv-isa-sim #467: support instruction misaligned senario

gxflying, 2020-05-20:
Hi, sir,  when I used the spike as a iss for our cpu,  i find that the behavior of instruction misaligned of spike is different fron our cpu.
our cpu will always treat the two LSB as 0 as a way to manage instruction misaligned.
while spike will always treat this as "instruction address misaligned".
so， I want to know whether we have a shedule to support this feature as a option?
thank~

scottj97, 2020-05-29:
The RISC-V spec requires an instruction address misaligned exception in this case. From the unprivileged ISA spec: "The JAL and JALR instructions will generate an instruction-address-misaligned exception if the target address is not aligned to a four-byte boundary." 

I don't think you should expect support from Spike for something that is not spec-compliant.

gxflying, 2020-05-30:
Yes, you are right as long as the spike is a iss of "riscv standard isa" simulator.
maybe I can implement this feature in my local code.
thank you any way~
and I will close this issue.

closed 2020-05-30 by gxflying