riscv-software-src/riscv-isa-sim #469: License issue with code scan

MDBrothers, 2020-05-21:
Hi guys,

Legal's code scanning software seems to think that some of the Spike code falls under GPL licenses.
To me it appears that everything is under at BSD license including softfloat and fdt.

Could I trouble the maintainers to make a statement about the Licenses? 

Right now I am tasked with painstakingly combing every code snippet the scanning software has flagged and refuting its determinations. It seems to be calibrated to avoid false negatives at all costs. 

Thanks again

aswaterman, 2020-05-21:
As the LICENSE file implies, Spike is intended to be BSD.

I wrote most of the code myself, so can speak for that.  `softfloat` isn't mine, but is clearly BSD-licensed.  The `fdt` code that @chihminchao recently added is dual-licensed under GPL and BSD, which might be the thing that's confusing your software.

MDBrothers, 2020-05-21:
The inputs for Autoconf are also getting flagged, but the text says that they take on the license of the overall program they are packaged with.

aswaterman, 2020-05-21:
Ah, it's annoying that doesn't work automatically, but I guess there's no legal issue.

MDBrothers, 2020-05-21:
I that's my opinion too. Thanks for the response.

closed 2020-05-21 by MDBrothers