riscv-software-src/riscv-isa-sim #471: Spike not working with helloworld MacOSx

yunusdawji, 2020-05-23:
When I launch 
$ spike pk hello

I get the following error

Assertion failed: (IS_ELF32(*eh64) || IS_ELF64(*eh64)), function load_elf, file ../fesvr/elfloader.cc, line 33.

Would you tell me why this could be?

aswaterman, 2020-05-23:
Probably unrelated to MacOS.

Sounds like your “hello” is not a valid ELF.

yunusdawji, 2020-05-23:
Thanks for reply. The binary work fine with the rtl (verilator) simulator. I this issue is related to spike or toolchain configuration. 

https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/lIWstAevafw

Maybe you can direct me with that if you have experience

yunusdawji, 2020-05-23:
I had a working setup before and I was able to run binary at rtl level. Now, something is missing.

I am using the toolchain installed via homebrew. And i compile a simple helloworld. Do you know where the issue could be or what checks should i perform?

yunusdawji, 2020-05-23:
Solved! There was a file name pk in that folder. And I guess that makes that it was complaining about invalid ELF for "pk" (not for "hello"). It would be good if the program throws the file name to avoid confusion.

closed 2020-05-23 by yunusdawji