riscv-software-src/riscv-isa-sim #478: Printing register values to trace

Rahul-Kande, 2020-06-05:
Hello, 

I am looking for a way to print register values into the trace. I know that if run in debug mode, I can print any register value that I want, but that will require me to enter the commands for every instruction. 

I want the value in the register that is updated by the current instruction. 
Ex: Core 0: 0x0000000080001000 0x00002be4 add x3, x1, x2  x3 = 0x2525a699   --> reg x3's value is printed since add wrote to x3

Is there a way to achieve this already? If not, can someone help me with this update? 

Thanking you in advance

aswaterman, 2020-06-05:
You can try configuring with `--enable-commitlog` and running with `--log-commits`

Rahul-Kande, 2020-06-05:
that totally worked. 

thank you very much.

closed 2020-06-05 by Rahul-Kande