riscv-software-src/riscv-isa-sim #479: Does multi-core in spike share same l1 cache?

XiaPZ, 2020-06-07:
Hi, I found in spike main, if there is multiple cores, these cores all register the same icache and dcache, maybe it does not quite make sense? because I think each core should have their own L1 cache while share the same L2 cache, am I missing something here?
https://github.com/riscv/riscv-isa-sim/blob/33a6eb57564c257037780ddd2691ca621c44a55b/spike_main/spike.cc#L375-L377

aswaterman, 2020-06-07:
Looks like you're right--it's effectively modeling the cache hierarchy of a multithreaded core, not a multicore.  I've prototyped a patch to model homogeneous private caches instead: https://github.com/riscv/riscv-isa-sim/pull/480

XiaPZ, 2020-06-08:
> Looks like you're right--it's effectively modeling the cache hierarchy of a multithreaded core, not a multicore. I've prototyped a patch to model homogeneous private caches instead: #480

thanks a lot

closed 2021-11-15 by XiaPZ