riscv-software-src/riscv-isa-sim #482: vle8.v checks alignment against rd

timsifive, 2020-06-09:
From the (modify) spike log:
```
core   0: 0x0000001212340224 (0x02050207) vle8.v  v4, (a0)
core   0: exception trap_illegal_instruction, epc 0x0000001212340224, due to is_aligned(insn.rd(), (*p).VU.vemul)
core   0:           tval 0x0000000000000000
```

Tracking the source:
vle8_v.h:
VI_LD(0, (i * nf + fn), int8);

decode.h VI_LD():
  VI_CHECK_LOAD(elt_width); \

decode.h VI_CHECK_LOAD():
  VI_CHECK_STORE(elt_width); \

decode.h VI_CHECK_STORE():
  require_align(insn.rd(), P.VU.vemul); \

That alignment check looks wrong to me, but I'm not confident enough in my understanding of the vector instruction set to know for sure. What do the experts think?

aswaterman, 2020-06-09:
It depends on what value of the `vtype` register.  If SEW=8 and LMUL=8 (i.e. vtype.vsew=0 and vtype.vlmul=3), then `rd` must be divisible by 8.

timsifive, 2020-06-12:
Got it. Thanks for pointing me in the right direction.

closed 2020-06-12 by timsifive