riscv-software-src/riscv-isa-sim #484: CXXABI_1.3.8/GLIBCXX_3.4.20 not found

DisturbedShadow, 2020-06-10:
Hello everyone, I am having a problem while trying to execute and simulate a file with spike. I am getting this error: 

spike: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by spike)
spike: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by spike)

I have searched in some posts to invoke the newest libraries to run with spike but it does not work.
My server gcc gnu version is 2014 but the riscv-toolchain is the latest one.

Thank you in advance

aswaterman, 2020-06-10:
Doesn't seem like a Spike issue.  Sounds like the kind of thing that happens when you compiled it in a different environment than you executed it in.

DisturbedShadow, 2020-06-10:
I've put the bin folder with all the necessary files into my PATH so I can execute them in a different folder, I guess this does not count?

aswaterman, 2020-06-10:
Seems like your runtime environment has an older libstdc++ version than your compile-time environment.  Sorry, I can't help more than that.

kito-cheng, 2020-06-10:
The easiest solution is recompile spike again on your environment, or recompile it with "-static-libstdc++".

DisturbedShadow, 2020-06-10:
How can I do that, I'm really new to everything.

DisturbedShadow, 2020-06-11:
What happens is that when I install spike on a different folder and not in riscv tools path then it works but then I can not use pk.

DisturbedShadow, 2020-06-11:
Solution:

If you want to execute spike everywhere like I wanted add spike to your PATH var then create an export var for the pk binary file (example: export pk=/path/to/your_pk_file) and if you want to use spike with rv32i isa, use it like this : spike --isa=rv32i $pk <your_program_for_sim.elf>

closed 2020-06-11 by DisturbedShadow