riscv-software-src/riscv-isa-sim #487: issue while using Half-precision Floats for riscv-v extension

soniab, 2020-06-12:
While trying to use Half-precision Floating point operations for riscv-v extension  instruction set architecture with spike.  My assembly code is giving me an error "An illegal instruction was executed!", when I am using vsetvli t0, a0, e32,m1,d2 followed by some operation like vfdot.vv or vfadd.vv in my assembly. 

As per my understanding,  Spike supports Half-Precision Floating point operations for riscv-v extension but I need to program using assembly programming at this stage. "Please correct me If my understanding is wrong". 

I will highly appreciate any help or pointer on the above issue.

chihminchao, 2020-06-12:
@soniab 
Do you specify half precision isa extension such as "--isa=rv64gcv_zfh"   ?

soniab, 2020-06-12:
Yes, I have specified "--isa=rv64gcv_zfh" but still getting the same error. Do I also need to specify isa extension as half precision while installing spike?

chihminchao, 2020-06-12:
The other possible cause is that mstatus.vs is not setup correctly.

soniab, 2020-06-12:
Is there any specific guidelines that you want me to follow while doing setup  for mstatus.vs?

chihminchao, 2020-06-12:
Just initialize it as non-zero value to enable vector unit  such what we have to do for floating unit

soniab, 2020-06-13:
I have verified value of mstatus_vs and it is set as non-zero value already. 
It is defined as "#define MSTATUS_VS          0x00000600" in riscv/encoding.h. 

But still I am getting the same error. 

In my assembly, If I have used following  vector instructions:
vsetvli a5, t0, e32,m1,d2
vfadd.vv        v1, v1, v2

After running with spike, I am getting error "An illegal instruction was executed!"

Do I need to set any other value or I am missing something. 

I will really appreciate your help in this regard.

chihminchao, 2020-06-14:
:vsetvli a5, t0, e32,m1,d2"  d2 means "EDIV 2" but current spike doesn't support ediv extension.

soniab, 2020-06-14:
Thanks for this update. This is very helpful. 

Would you please help me, which instructions I can use for half-precision floating point operations in my assembly for riscv-v extension. 
Because float16_t, I am not able to use it as data type in my test program. With float (32b) as data type in my test program, I thought only this way to use for half-precision floating point instructions in assembly.

chihminchao, 2020-06-15:
You may ask for correct gcc and binutils version in compiler or riscv sw-dev google group.  I am not sure about their status.