riscv-software-src/riscv-isa-sim #492: What is the option of vector isa when running spike ? --isa=rv32iv

WilliamWangPeng, 2020-06-18:
spike --isa=rv32im ./rv64/riscv-tools/riscv-gnu-toolchain/riscv32-unknown-elf/bin/pk ./test/t1    
where t1 is a 32 bit elf which can be seen using readelf -h t1, pk is also choosen 32 bit part, I'm compiling a 32 bit vector instruction vsetvl,  do you know which --isa=rv32iv is useful?  

[image]  
  
thank you 
Best Regards
William

chihminchao, 2020-06-18:
1. "bad --isa option rv32v",   i extension is at least required 
2. rv32iv should work if you only touch integer vector instruction.
3. for bbl loder illegal instruction,  insn (0000862e) is "c.mv a2, a1).  It means c extension is required or the problem should be compiled without c extension.

WilliamWangPeng, 2020-06-18:
> 1. "bad --isa option rv32v",   i extension is at least required
> 2. rv32iv should work if you only touch integer vector instruction.
> 3. for bbl loder illegal instruction,  insn (0000862e) is "c.mv a2, a1).  It means c extension is required or the problem should be compiled without c extension.

Hi dear Chao
thanks for reply.   
```    
spike  --isa=rv32i  pk hello  
```      
 works, but when I choose   
```    
spike  --isa=rv32i  pk t2  
```    
where t2 is a 32 bit RISCV elf and contains the vsetvl intruction at t2.s firstly.  
error: bad --isa option rv32iv  

[image]

chihminchao, 2020-06-18:
vector unit must be enabled by mstatus.vs before using any of its instructions.
Do you check that ?

WilliamWangPeng, 2020-06-18:
> vector unit must be enabled by mstatus.vs before using any of its instructions.
> Do you check that ?

oh thank you Chao, I didn't do that  
"vector unit must be enabled by mstatus.vs before using any of its instructions."  
how can I enable mstatus.vs before using any of its instructions

WilliamWangPeng, 2020-07-08:
> vector unit must be enabled by mstatus.vs before using any of its instructions.
> Do you check that ?

hi dear chao, how can I enable mstatus.vs before using any of its instructions

closed 2020-08-03 by WilliamWangPeng