riscv-software-src/riscv-isa-sim #497: printf with %f and %g flags not showing correct output

nagendragd, 2020-06-29:
I have the latest master branch of riscv-isa-sim cloned.

A simple printf("hello world %g\n", val) where val=3.1415; generates the incorrect output
"hello world g"

Tried replacing %g with %f and again it produces incorrect output "hello world f"

However %d works correctly.

I am compiling the binary using gcc with the flags -march=rv32imafdc

I have configured pk (process kernel) with the same flags.

I am running spike with the same flags specified to the --isa argument.

This used to work correctly in a previous version of spike, from a couple of months ago or so.

Any clues will be much appreciated.

Thank you
Nagendra

aswaterman, 2020-06-30:
This has nothing to do with Spike.  You are probably now using a C library that doesn't support floating-point in stdio, whereas lsat time, you were using a C library that did support that feature.

nagendragd, 2020-06-30:
Oh, ok. I will investigate the C lib issue. Thanks for sorting me out..

closed 2020-06-30 by nagendragd