riscv-software-src/riscv-isa-sim #498: Spike error Make

Fifiy, 2020-07-02:
Hello, 
I was trying to add a new instruction to spike, but when I rebuild it I get this error as below saying: undefined reference to rv64_mod(processor_t*, insn_t, unsigned long). I've checked my Path, but can't recognise where this is coming from ! Thanks.

[image]

chihminchao, 2020-07-02:
It is usually that the name in riscv.mk.in doesn't match the file name in riscv/insns/the_instruction_name.h.  You may check that.

Fifiy, 2020-07-02:
Actually they are the same !
[image]
[image]

WilliamWangPeng, 2020-07-08:
> Actually they are the same !
> [image]
> [image]

You need to focus on the format in riscv_insn_ext_i= \
mod and \ shall have a space