riscv-software-src/riscv-isa-sim #503: The  assembler representation of a "xor zero, zero, zero" instruction is missing in a log file.

dariuszst, 2020-07-09:
The log file generated for a code:
```
.globl _start
.section .text
_start:
**xor zero, zero, zero**
xor ra, ra, ra
_end:
j _end
_exit:
```
contains the following entries:

...
***core   0: 0xffffffff80000000 (0x00004033)  - 
3 0x80000000 (0x00004033)***
core   0: 0xffffffff80000004 (0x0010c0b3) xor     ra, ra, ra
3 0x80000004 (0x0010c0b3) x 1 0x00000000
...

The bolded lines represent a xor instruction involving zero register used three times. While it looks that simulation is performed properly and the binary representation (0x00004033) of this instruction looks OK then  the assembler representation is missing. 
The ovpsim prints the proper log:

...
***Info 1: 'riscvOVPsim/cpu', 0x0000000080000000(_start): Machine 00004033 xor     zero,zero,zero***
Info 2: 'riscvOVPsim/cpu', 0x0000000080000004(_start+4): Machine 0010c0b3 xor     ra,ra,ra
...

Is it any reason that assembler code is missing or it is just a bug?

sorear, 2020-07-22:
the Spike disassembler is also used for making hardware commitlogs human-readable, and some UCB hardware uses `xor x0,x0,x0` to represent a pipeline bubble, so it is suppressed in the output.

https://github.com/riscv/riscv-isa-sim/blob/master/spike_main/disasm.cc#L492
https://github.com/ucb-bar/riscv-sodor/blob/e4d10a117ca5ac81a806703dc375ceed7f8a37d0/src/common/consts.scala#L27

aswaterman, 2020-07-22:
I had forgotten about that ancient esoterica.  It predates the standardization of the HINT encoding space, and `xor x0, [anything]` is currently reserved for standard use.  So I don't think it's appropriate for Spike to endorse it anymore.

commit 67b7edd0 (2020-07-22): Remove deprecated decoding of xor x0,x0,x0

Some UCB implementations once used this to represent a pipeline bubble.
But this encoding is reserved for future standard HINT use.

Resolves #503
closed 2020-07-22 by aswaterman