riscv-software-src/riscv-isa-sim #505: cann't run vector ISA on riscv-linux

ALLap97, 2020-07-14:
I ran V ISA on spike and pk successfully.
```
riscv64-unknown-linux-gnu-gcc -c callsaxpy.c -o callsaxpy.o
riscv64-unknown-linux-gnu-as -march=rv64gcv --mabi=lp64d saxpy.s  -o saxpy.o
riscv64-unknown-linux-gnu-gcc -static saxpy.o callsaxpy.o -o out
spike --isa=RV64GCV pk out

bbl loader
1.000000
2.000000
3.000000
----------------
2.000000
4.000000
6.000000

```
>
and i ran riscv-linux on spike  successfully, and i can run G ISA elf on riscv-linux.
but i got some problem  when  i tried to run the same rvv_elf on riscv-linux.

`
spike  --isa=RV64GCV riscv-pk/build/bbl `
>
in riscv-linux(busybox)
`
./riscv_rvv
`
>
error

```
./riscv_rvv
1.000000
2.000000
3.000000
----------------
[    3.630000] riscv_rvv[46]: unhandled signal 4 code 0x1 at 0x000000000001040c in riscv_rvv[10000+59000]
[    3.630000] CPU: 0 PID: 46 Comm: riscv_rvv Not tainted 4.20.0-44631-g8fe28cb58bcb #8
[    3.630000] sepc: 000000000001040c ra : 00000000000104f2 sp : 0000003fffee2c40
[    3.630000]  gp : 000000000006c110 tp : 000000000006e710 t0 : 0000000000000000
[    3.630000]  t1 : 000000000001d600 t2 : 0000000000000000 s0 : 0000003fffee2c80
[    3.630000]  s1 : 0000000000010aaa a0 : 0000000000000003 a1 : 0000003fffee2c50
[    3.630000]  a2 : 0000003fffee2c40 a3 : 5420000000000000 a4 : 0000003fffee2c40
[    3.630000]  a5 : 0000003fffee2c50 a6 : 2d2d2d2d2d2d2d2d a7 : 0000000000000040
[    3.630000]  s2 : 0000000000000000 s3 : 0000000000135218 s4 : 000000000018bbc0
[    3.630000]  s5 : 0000000000000008 s6 : 000000000018a798 s7 : 000000000018bb88
[    3.630000]  s8 : 0000000000000000 s9 : 0000000000135bb0 s10: 0000003fffdcefa4
[    3.630000]  s11: 0000000000135bf0 t3 : 0000000000000000 t4 : 0000000000000000
[    3.630000]  t5 : 0000003fffee24a8 t6 : 0000000100000000
[    3.630000] sstatus: 8000000200006020 sbadaddr: 0000000000b57757 scause: 0000000000000002
Illegal instruction

```

aswaterman, 2020-07-14:
Linux doesn't currently support the vector extension.

ALLap97, 2020-07-14:
i donn't understand the difference between using pk and linux.
Isn't all instructions run through spike?

aswaterman, 2020-07-14:
Yes, but the OS kernel needs to understand and support the additional ISA extension.

kito-cheng, 2020-07-14:
mstatus needs to initialize before execute vector instructions, pk will do
that[1], but linux kernel won't do that
since it is unsupported yet, and I guess it's impossible to initialize that
in your program because
mstatus register require M-mode to write.

https://github.com/riscv/riscv-pk/commit/77a5df569451571d608650a34183d53df99790ec

kito-cheng, 2020-07-14:
Oh and that's not the only thing needed to support vector extension, you
also need to handle vector register and related CSR during content switch,
PK only supports single thread/single process, so isn't a problem when you
run program with PK.

kito-cheng, 2020-07-14:
In case you are interested on linux kernel support:
https://patchwork.kernel.org/project/linux-riscv/list/?series=301701

ALLap97, 2020-07-14:
OK！
thanks a lot.

closed 2020-07-15 by ALLap97