riscv-software-src/riscv-isa-sim #509: Debugging with GDB

kingstone1927, 2020-07-17:
[image]
[image]

I followed the "Debugging With GDB" instructions from the README and got these warnings and errors when running spike and openOCD. I'm pretty sure that I followed exactly what was in the instruction. 

Thanks in advance

timsifive, 2020-07-17:
Nothing looks out of place.

Does it work?

kingstone1927, 2020-07-18:
> Nothing looks out of place.
> 
> Does it work?

Actually, it does work. I just wonder what are those errors and warning? Also, how can I learn to write a linker file (.lds) and configure file (.cfg)?

Thanks much