riscv-software-src/riscv-isa-sim #511: Issue: vfmv_v_f intrinsics with half precision

soniab, 2020-07-20:
Dear Team,

While using   vfmv_v_f (broadcast) intrinsic with half precision with spike, I observed that this intrinsic is always broadcasting NaN value. To debug this issue, I reviewed "isBoxedF16()" function present inside riscv/decode.h and found an issue that this function is always returning false value. 

Therefore, to fix this issue I made following changes in  "isBoxedF16" function:
#define isBoxedF16(r) (isBoxedF32(r) && ((uint64_t)((r.v[1] >> 16) + 1) == ((uint64_t)1 << 48)))

I verified this change and now I got correct results while using vfmv_v_f intrinsic for half precision in my application. I did test some other scenarios also to verify this change and got correct results.

aswaterman, 2020-07-20:
The current definition seems correct: `r.v[1]` should be all ones, whereas the 48 high bits of `r.v[0]` should be all ones.  Your version ends up being equivalent to `isBoxedF32(r)`, i.e., `((uint64_t)((r.v[1] >> 16) + 1) == ((uint64_t)1 << 48))` is true whenever `isBoxedF32(r)` is true.

I'm wondering if for some reason the input to your intrinsic does not have all of bits XLEN-1..16 == 1?

cc @chihminchao

soniab, 2020-07-21:
Hi,

For using intrinsics for half precision, I am doing following steps : 
1) Convert float (32bit) to float16 (16 bit). 
     for ex. -  float alpha = 4.5. convert alpha to float16_t alpha16
2) use intrinsic (vfmv_v_f) on alpha16 (float 16bit)

After following above steps, I was getting NaN value with vfmv_v_f intrinsinc. 
So, I am not sure if all bits for alpha16 needs to be equals to 1 to broadcast any input. 

Please let me know if my understanding is correct.

chihminchao, 2020-07-21:
@soniab 
I also don't think it is reasonable modification.  For an valid NaN-Boxing float16, r.v[1] should be all one's.
You mentions you uses intrinsic. 
Do you means use gcc intrinsic instruction in C to manipulate vector unit or something else ?
Could you provide snippet of your code ?

aswaterman, 2020-07-21:
It seems the conversion from float32 to float16 is failing to NaN-box the result (i.e. fill all of bits XLEN-1..16 with 1s).

soniab, 2020-07-22:
I am hereby sharing sample code. It contains three files. 
1) test_fill.c : this file converts float to float 16 and then call fill() function available in assembly program. 
2) fill1.s: contains fill() function to use riscv-v extension intrinsics to broadcast the value and store the same in an array. 
3) output: contains the output which I am getting.

[sample-code.tar.gz](https://github.com/riscv/riscv-isa-sim/files/4959706/sample-code.tar.gz)

I am using softfloat library functions to convert float to float16 and vice-versa. 
I am using gcc compiler from riscv-gnu-toochain. 

Please let me know if you need any further details.

aswaterman, 2020-07-22:
We can't help you debug your code.  It seems to us your code is failing to NaN-box its float16 results.  You might need to do this manually; SoftFloat won't do it for you.