riscv-software-src/riscv-isa-sim #514: ZFH Extension, is it proprietary? Can't find it.

MDBrothers, 2020-07-23:
The ZFH extension doesn't appear to be in the libcustomext.so in the 7/01/2020 version of the master branch.

There does not appear to be a ZFH related so that gets built.

Is that because this is a proprietary extension?

If so is there a way to negotiate to allow Futurewei Technologies Inc access to this?

aswaterman, 2020-07-23:
Not proprietary. It’s being standardized through the risc-v foundation.

MDBrothers, 2020-07-23:
That's good news.

Is it just not being distributed yet or did I misconfigure something?

aswaterman, 2020-07-23:
This branch of the ISA spec includes it. https://github.com/riscv/riscv-isa-manual/tree/zfh

MDBrothers, 2020-07-23:
Ok thanks for your help!

MDBrothers, 2020-07-23:
Forgive me if I'm missing something, but it appears that repo only has tex files.

aswaterman, 2020-07-23:
Yep.  If you don't have latex installed, you can use the PDF I've attached here.

[zfh.pdf](https://github.com/riscv/riscv-isa-sim/files/4969630/zfh.pdf)

closed 2020-07-23 by aswaterman

MDBrothers, 2020-07-24:
You misunderstand me. I know how to build a pdf from a tex file.

**Spike has code in it that goes looking for libzff.so if you specify the extension _zfh.**

I want the **library**. The library is not seemingly available.

I just want an answer if the library is meant to be built by spike, included with or what.

I do not need additional documentation.

kito-cheng, 2020-07-24:
I thought zfh should included in master branch?
https://github.com/riscv/riscv-isa-sim/pull/489

$ spike --isa=rv64gc_zfh

MDBrothers, 2020-07-24:
Yes that is the isa string I am using.

I added an API to Spike. 

I'm thinking nobody understands my question because zfh "just works" for users of vanilla Spike, but I may have changed something in a way to disrupt the process by which extensions are validated.

MDBrothers, 2020-07-24:
Yeah, sorry guys, that was exactly what happened. 

My version of processor.cc diverged and when merging the upstream changes I had accidentally left in an extra call to register_extension.

Sorry for the confusion and my frustration. I truly do appreciate the effort of the Spike developers and their willingness to respond to questions.

aswaterman, 2020-07-24:
Great - glad you figured it out!