riscv-software-src/riscv-isa-sim #516: Masked vslidedown

Marc43, 2020-07-27:
Hi,

in the 0.9 vector specification `vslidedown` behaviour is:

[image]

There's a "masked" behaviour, my question is, is this implemented in the newest version of spike?

Looking at the code it seems like not, variable `is_valid` only takes into account the current `position`,  `offset` and `vlmax`.

Thanks!
Marc

aswaterman, 2020-07-27:
The `VI_LOOP_BASE` macro skips masked loop iterations.

Marc43, 2020-07-27:
Okay, missed that one, sorry and thanks :)

aswaterman, 2020-07-27:
cool, glad to help.

Marc43, 2020-07-27:
Sorry @aswaterman for reopening this, but, is enough with skipping the iterations? Or should the value be written to 0 explicitly?

I tell you this because I am using Spike as a reference model and the values differ because the upper elements (with the mask enabled for those) are not set to 0 in Spike.

aswaterman, 2020-07-27:
Can you provide a concrete example that violates the spec?  I'll let @chihminchao handle it from here.

Marc43, 2020-07-27:
Sorry again for bothering you. 

I think I misunderstood how it is programmed. I understand that if the mask is not enabled (== 0), the iteration is skipped as the element must be unchanged. When mask is enabled (!= 0), the rest of the behaviour is checked and the element is written.

closed 2020-07-27 by Marc43