riscv-software-src/riscv-isa-sim #522: `mie.mtie` enabled resulting in unexpected traps

udinator, 2020-08-06:
Hi all,

I'm seeing something strange occur when running a randomly generated program on Spike.

If I enable all interrupts except for timer interrupts, then Spike correctly executes the program and reaches the end without any issue.
However if I enable timer interrupts by setting `mie.mtie/stie/utie`, then Spike will execute exactly 5000 instructions and then suddenly trap to handle a timer interrupt, as shown below:
[image]
I'm not sure why this is happening, could anyone provide some guidance on how to resolve this issue?

Thanks!

Note: the command I'm running is `<spike_path>/spike --log-commits --isa=rv32imc -l <asm_binary>`

aswaterman, 2020-08-06:
Spike only checks for interrupts periodically (every 5000 instructions in
most cases).  The timer interrupt probably became pending earlier (the
condition, recall, is mtimecmp <= mtime), but Spike was just a bit tardy in
taking the interrupt.

udinator, 2020-08-20:
Understood.
Is there any way to prevent Spike from taking these sorts of internal "watchdog" interrupts?
I'm using Spike logs to check functional correctness against a core that does not natively implement `mtime` and `mtimecmp` (they are provided by the platform instead).

aswaterman, 2020-08-20:
The standard approach would be to set `mtimecmp` to -1 during the boot sequence, so that the interrupt won't occur for 2^64 (a.k.a. infinity) cycles.

If that doesn't work for you for some reason, you could make a one-off hack to Spike to remove the code that sets mip.mtip.