riscv-software-src/riscv-isa-sim #525: RVV Load and LMUL issues

bbougenot, 2020-08-10:
Hello,

I'm having trouble getting basic vector instructions to run on spike. Initially I was using spike from commit 7246912cd7e21f0c68cd3258ae4ab59e4d836155 and all my tests worked except for when I would try to use a LMUL > 1. As soon as LMUL is increased spike triggers:
"exception trap_illegal_instruction, epc 0x000000008000277c
core   0:           tval 0x0000000000000000"
on the instruction vle64.v v1, (a5). I thought perhaps LMUL wasn't fully supported in this version of spike so I upgraded to the latest commit but now every single load instruction triggers this trap regardless of whether LMUL is 1 or not. I haven't changed anything from my code. For reference this is what I'm running:

Initialize the vector extension:
`  __asm__ volatile (    \
  "csrr t0,mstatus \n"  \
  "li  t1,0x600 \n"     \
  "or t1,t1,t0 \n"      \
  "csrw mstatus,t1 \n"  \
  );  `

Set the vector configuration (In this case VLEN is 5):
`__asm__ volatile ("vsetvli t0, %[A], e64, m1" :: [A] "r" (VLEN));  `

Allocate a buffer to load:
` volatile uint64_t BUFF[] = {1, 2, 3, 4, 5}; `

Load the vector:
`__asm__ volatile ("vle64.v v1, 0(%0)":: "r" (BUFF)); `

Any help is much appreciated.

aswaterman, 2020-08-11:
This is likely because Spike has moved onto the RVV v1.0 draft, which has reorganized the layout of the vtype register.  You'll need to update to an RVV v1.0 assembler.  (Perhaps @chihminchao can provide you a link.)

bbougenot, 2020-08-12:
Hey @aswaterman, that makes sense. I noticed the vsetvl instruction in the debug mode of spike didn't correspond to the same instruction dumped from the binary. However, I still don't understand why I can't get LMUL > 1 to work on the spike version for 0.9 of the riscv V vector extension. I always get trap illegal instruction for loads/stores except for when I used LMUL = 1. Any ideas?

aswaterman, 2020-08-12:
Same reason - upgrade your toolchain.