riscv-software-src/riscv-isa-sim #529: Have to do overlap checking in VRGATHEREI16.vv

Rolfy360, 2020-08-14:
The current spike validates the relations among rd, rs2, rs1 using

if (__builtin_expect(!(insn.rd () != insn.rs2 () && insn.rd () != insn.rs1 ()), 0))
throw trap_illegal_instruction (0);

rd and rs1 should use is_overlapped(.) to check the overlap condition.

case: 
for VLEN=128, VSEW=64, VMUL=4
vrgatherei16 v8, v4, vs1
vs1 contains 8 e16 elements(fits in 1 VPR) and can not be v8, v9, v10, v11.

aswaterman, 2020-08-14:
@chihminchao I did not test this, but it is true that vs2 and vs1 cannot overlap rd, and vs1 needs special handling for this instruction because its EMUL might be less, equal, or greater than vd.

chihminchao, 2020-08-20:
@Rolfy360 
Thanks for you report.  The fix is at https://github.com/riscv/riscv-isa-sim/pull/533

aswaterman, 2020-08-20:
Closed via #533

closed 2020-08-20 by aswaterman