riscv-software-src/riscv-isa-sim #531: Any example about how to use this vector extension in spike?

fanghuaqi, 2020-08-19:
Hi, I tried to write a test code based on https://github.com/riscv/riscv-v-spec/blob/master/example/vvaddint32.s, and compiled it using riscv gcc compiled using branch rvv-0.9.x, and tried to run it usin spike, it reported ilegal instruction on `vsetvli t0, a0, e32, ta,ma`,  I tried to run the program using `spike --isa rv32gc demo_vpu.elf`.

The spike is configured using `../configure --prefix=/home/hqfang/mysofts/riscv32 --with-isa=RV32IMAFDC --enable-misaligned --enable-commitlog --enable-histogram`

Thanks
Huaqi

chihminchao, 2020-08-19:
please use "--isa rv32gcv"

fanghuaqi, 2020-08-19:
I also tried to use `spike --isa rv32gcv demo_vpu.elf`, but it failed at another instruction `vle32.v v0, (a1)         # Get first vector`.

Thanks
Huaqi

chihminchao, 2020-08-20:
the mstatus.vs field needs to be setup non-zero value to enable vector unit.  Just like mstatus.fs for floating-point unit

fanghuaqi, 2020-08-20:
Ok， I also have question that I tried to debug in spike, but the behaviour in spike is strange, the code below:
~~~
    srli t0, a0, 1
    vsetvli t0, t0, e64, m8 # Set vector length based on 64-bit vectors
    slli t0, t0, 1
    sub a0, a0, t0         # Decrement number done
~~~
is translated in spike as below:
~~~
core   0: 0xffffffff80000180 (0x00155293) srli    t0, a0, 1
: 
core   0: 0xffffffff80000184 (0x00f2f2d7) vsetvli t0, t0, e16, mf2, tu, mu
: 
core   0: 0xffffffff80000188 (0x00000286) c.slli  t0, 1
: 
core   0: 0xffffffff8000018a (0x40550533) sub     a0, a0, t0
: 
core   0: 0xffffffff8000018e (0x00000001) c.nop
~~~

The `vsetvli` is not recognized correctly, command I use is `spike --isa=rv32imacv --varch=vlen:128,elen:64,slen:128 -d demo_vpu.elf`.

Here is my code:

[demo_vpu.zip](https://github.com/riscv/riscv-isa-sim/files/5101475/demo_vpu.zip)

I have already enabled mstatus.vs in main function, before any operation on vector instructions.

Thanks
Huaqi

aswaterman, 2020-08-20:
The layout of the `vtype` register changed between RVV 0.9 and RVV 1.0.  It looks like you're using a recent version of Spike with an old version of the assembler.

fanghuaqi, 2020-08-20:
I am using gcc toolchain built by https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-0.9.x. And using the latest spike to run.

aswaterman, 2020-08-20:
Right - those are mismatched.  You can try using an older version of Spike (perhaps 67b7edd0270d5d074136eb45474477d15aeee725)

fanghuaqi, 2020-08-20:
Thanks for your information, it helps for me @aswaterman  @chihminchao , I used the spike version https://github.com/riscv/riscv-isa-sim/commit/67b7edd0270d5d074136eb45474477d15aeee725 it works.

closed 2020-08-24 by fanghuaqi

wangliwei-intel, 2022-01-05:
@aswaterman 
Hi, Current LLVM/Clang only supports RVV-V0.10, to run code built from it, which commit of spike should I choose?
Thanks.

aswaterman, 2022-01-05:
RVV 0.10 and 1.0 are essentially fully compatible, so the HEAD of the master branch should suffice.

wangliwei-intel, 2022-01-05:
@aswaterman 
Thank you.
My experiment also justified your point, the code (RVV-V0.10) could run against master branch without any exception.
But the computation result is wrong, I'm wondering what gent wrong? Do you have any idea about it?

aswaterman, 2022-01-05:
No idea, and I can't offer help debugging it.  But I'm sure RVV 0.10 vs 1.0 is not the reason.

wangliwei-intel, 2022-01-05:
@aswaterman 
Got it, thank you, good day.

wangliwei-intel, 2022-01-17:
Hi,
When I type 'freg 0 fa0' for showing content of fa0, I got this: 0xffffffffffffffffffffffff3dcccccd.
It's 128-bit long instead of 64-bit, Did I do something wroing?

When I input 'fregd 0 fa0', it shows me "-nan" while the value is supposed to be '0.1'.
What's wroing with it?

Thanks.
Liwei

aswaterman, 2022-01-17:
There are two unrelated issues here:
- Spike's internal implementation has 128-bit FP registers to support the Q extension, and so the debug mode prints out 128 bits, even when the Q extension is disabled.  You can safely ignore the upper 64 bits.
- NaN is the correct result: `ffffffff3dcccccd` is a double-precision NaN.  Maybe you meant to write `fregs`?  `3dcccccd` is inded approximately 0.1 in single-precision.

wangliwei-intel, 2022-01-17:
oh, my bad, I should use 'fregs' instead.
Thank you, @aswaterman