riscv-software-src/riscv-isa-sim #536: Spike spin in the loop before exit

Phantom1003, 2020-08-28:
These days I have a look on spike log file, I find that spike spin in the loop before exit the program, like following

```
3 0x00001000 (0x00000297) x 5 0x00001000
3 0x00001004 (0x02028593) x11 0x00001020
3 0x00001008 (0xf1402573) x10 0x00000000
3 0x0000100c (0x0182a283) x 5 0x80000000
... work well
3 0x80000010 (0x00900f93) x31 0x00000009
3 0x80000014 (0x03ff0663)
3 0x80000018 (0x00b00f93) x31 0x0000000b
3 0x8000001c (0x03ff0263)
---- Notice ----
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
3 0x80000048 (0xff9ff06f)
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
3 0x80000048 (0xff9ff06f)
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
3 0x80000048 (0xff9ff06f)
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
3 0x80000048 (0xff9ff06f)
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
... loops thousands time
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
3 0x80000048 (0xff9ff06f)
3 0x80000040 (0x00001f17) x30 0x80001040
3 0x80000044 (0xfc3f2023) mem 0x80001000 0x00000001
```

I use rv32mi-p-csr to test, the code in 0x80000040 - 0x80000048 is trying to write to tohost
```
80000040 <write_tohost>:
80000040:	00001f17          	auipc	t5,0x1
80000044:	fc3f2023          	sw	gp,-64(t5) # 80001000 <tohost>
80000048:	ff9ff06f          	j	80000040 <write_tohost>
```

I guess the reason for this is because fesvr needs to wait until it is scheduled to be able to set the termination of spike, is this understanding correct, thanks

aswaterman, 2020-08-29:
Yeah, to improve the simulator's performance, the HTIF code only executes every few thousand cycles.  This is controlled by the constant `INTERLEAVE` in `sim.cc`, but it isn't really meant to be adjusted by the user; things like LR/SC loops and timer interrupts will break if you make the value too small.

Phantom1003, 2020-08-29:
Thank you, in normal mode the step is INTERLEAVE 
https://github.com/riscv/riscv-isa-sim/blob/5f76a0d1fa68bb80560cb890405c42041f744e89/riscv/sim.cc#L116-L131
But, if we go to the sim_t::step
https://github.com/riscv/riscv-isa-sim/blob/5f76a0d1fa68bb80560cb890405c42041f744e89/riscv/sim.cc#L140-L157
I find that spike will switch to host after every step, I think I have a misunderstanding about the `switch_to`, if the host.switch_to means go back to htif, spike should immediately stop when to_host is set to 1

closed 2020-09-16 by Phantom1003