riscv-software-src/riscv-isa-sim #538: Load/store address breakpoint priority

scottj97, 2020-09-01:
If a load instruction goes through [`load_slow_path()`](https://github.com/riscv/riscv-isa-sim/blob/f974ce1c6d30250d46d2b763715724bcd4b7a89d/riscv/mmu.cc#L152) it appears it will prioritize load access fault above a load/store address breakpoint, which is contra the priority list in privileged ISA spec table 3.7.

aswaterman, 2020-09-01:
@timsifive would have to comment, since he wrote the code, but presumably the explanation is that it's impossible to give data triggers higher priority than access exceptions (since you can't trigger if you don't have the data).

scottj97, 2020-09-02:
Perhaps the same issue applies to instruction fetch, since instruction address breakpoint is supposed to be the highest priority exception.

aswaterman, 2020-09-02:
Yeah.

I guess it will require some refactoring to separate the address and data trigger checks so they can be put in the right places.

TracyCai-16, 2021-02-24:
Any update on this?
Seems only address trigger is supported by now.
Will the priority issue get fixed recently?

scottj97, 2023-01-15:
Fixed by #1105

closed 2023-01-15 by scottj97