riscv-software-src/riscv-isa-sim #539: Data triggers on misaligned load/store

scottj97, 2020-09-01:
If I set a data trigger (type=2 select=1), a misaligned load/store will match incorrectly, since `processor_t::trigger_match()` will get called once per byte, instead of once with the entire loaded data. See [`mmu_t::misaligned_load()`]( https://github.com/riscv/riscv-isa-sim/blob/f974ce1c6d30250d46d2b763715724bcd4b7a89d/riscv/mmu.h#L66).

The current structure also precludes implementing the optional size field in mcontrol since the trigger_match routine doesn't know the size of the access.

aswaterman, 2020-09-01:
It's not obvious to me this should be considered a bug, since machines that trap and emulate misaligned accesses would also match on bytes.  But if you conclude it is a bug, you can make a PR to fix it.

scottj97, 2020-09-02:
Interesting point...I can also imagine hardware misaligned implementations that match on each half of the data separately.

scottj97, 2020-09-04:
Closing since the expected behavior is unclear, the current behavior seems somewhat reasonable, and I don't care.

closed 2020-09-04 by scottj97

scottj97, 2020-09-07:
> I think the behavior you're describing is that if I set a 1-byte trigger on a read of 0x1234, that trigger will fire when the code performs a 32-bit read at 0x1233. Ideally it wouldn't, since the code didn't do an 8-bit read.

I was referring to a data trigger (mcontrol.select=1). If I do an unaligned 32-bit load, Spike will compare the four bytes of data individually vs tdata2, instead of the entire 32-bit quantity written into the destination register.

But I suspect you're right, there's also an issue with the sizelo/sizehi fields when matching the size of the load.