riscv-software-src/riscv-isa-sim #540: Maximum number of physical address bits?

kathlenehurt-sifive, 2020-09-01:
I'm getting mixed messages about the maximum physical address supported by RV64. In the ISA, it states that RV64 provides full 64-bit physical address space. However in Spike, I see a max physical address of 56-bits always being enforced which is the maximum allowed in Sv48 mode. 

From `riscv/sim.cc`:
```
  static bool paddr_ok(reg_t addr)                                        
  {                                                                             
    return (addr >> MAX_PADDR_BITS) == 0;           
  }     
```

I have a test that reads and writes to addresses with these upper 10-bits set in machine mode (so no translation) and it runs fine on Spike. I just want to understand why it works.

aswaterman, 2020-09-01:
Sv39 and Sv48 virtual memory schemes impose a 56-bit limit on physical addresses.

There is a test that relies on bit 63 being set causing an access exception (https://github.com/riscv/riscv-tests/blob/a5d8386eb08cd47df56bf8170bcaf9e1f9b92c13/isa/rv64mi/access.S#L18), and it passes on Spike, so I'm surprised by the behavior you're seeing.  LMK if you figure out why it's happening.

kathlenehurt-sifive, 2020-09-02:
> Sv39 and Sv48 virtual memory schemes impose a 56-bit limit on physical addresses.

Does this limit apply to physical addresses accessed when translation is disabled? Like in machine mode? If so, it's unclear in the ISA what constitutes an illegal access when translation is disabled. Can you clarify this?

I ran the `rv64mi-p-access` test you mentioned and do see the load access fault being triggered (as expected):
```
  core   0: 0x00000000800001cc (0x00028383) lb      t2, 0(t0)
  core   0: exception trap_load_access_fault, epc 0x00000000800001cc
  core   0:           tval 0x80000000800001dc
```

It took some debugging, but I figured out my problem. I assumed since I was executing in machine mode that translation would be disabled. I didn't realize that translation could be enabled in machine mode under certain conditions:
> The MPRV (Modify PRiVilege) bit modifies the privilege level at which loads and stores execute in all privilege modes. When MPRV=0, loads and stores behave as normal, using the translation and protection mechanisms of the current privilege mode. When MPRV=1, load and store memory addresses are translated and protected, and endianness is applied, as though the current privilege mode were set to MPP. Instruction address-translation and protection are unaffected by the setting of MPRV. MPRV is hardwired to 0 if U-mode is not supported.

aswaterman, 2020-09-03:
> Does this limit apply to physical addresses accessed when translation is disabled?

Technically, this might be platform-defined at the moment, but in the platform Spike models, the limit is in effect.

> It took some debugging, but I figured out my problem.

Cool!  I guess I can close the issue.

closed 2020-09-03 by aswaterman

sorear, 2020-09-03:
IMO a requirement that high physical address bits be 0 is functionally the same as a PMA denying all access to addresses above a limit, and PMA checks are not translation and are active in all modes.

aswaterman, 2020-09-03:
In that spirit, Spike's PMAs are such that physical addresses >= 2^56 are invalid.