riscv-software-src/riscv-isa-sim #541: Spike does not run 32bit binaries in 32bit mode.

mattwachter, 2020-09-07:
I am trying to go through your tutorial [Debugging with Gdb](https://github.com/riscv/riscv-isa-sim#debugging-with-gdb) with 32bit binaries instead of 64bit.

```
spike --rbb-port=9824 -m0x10000000:0x20000 rot13-32 --isa=RV32IMAFDC
```

Connecting to the remote target in `gdb` fails with the error "fd requires xlen 4, but target has xlen 8".

The reason for this is probably that I cannot get `spike` to simulate binaries in 32bit mode.
When I simulate the binary in `spike` directly all register contents are displayed as 64bit values:
```
spike pk rot13-32 --isa=RV32IMAFDC

: pc 0
0x0000000080000008
: reg 0
zero: 0x0000000000000000  ra  : 0x0000000080000138  sp  : 0x0000000000000000  gp  : 0x0000000000000000  
tp  : 0x0000000000000000  t0  : 0x0000000000000000  t1  : 0x0000000000000000  t2  : 0x0000000000000000  
s0  : 0x0000000000001020  s1  : 0x0000000000000000  a0  : 0x00000000800001b0  a1  : 0x0000000000000002  
a2  : 0x00000000800001c0  a3  : 0x0000000000000000  a4  : 0xffffffffd00dfeed  a5  : 0x0000000000000000  
a6  : 0xffffffffff00ff00  a7  : 0x0000000000000000  s2  : 0x0000000000000000  s3  : 0x0000000000000000  
s4  : 0x0000000000000000  s5  : 0x0000000000000000  s6  : 0x0000000000000000  s7  : 0x0000000000000000  
s8  : 0x0000000000000000  s9  : 0x0000000000000000  s10 : 0x0000000000000000  s11 : 0x0000000000000000  
t3  : 0x0000000000000000  t4  : 0x0000000000000000  t5  : 0x0000000000000000  t6  : 0x0000000000000000
```

This is how I compiled the `rot13-32` binary:
```
riscv32-unknown-elf-gcc  -g -Og -o rot13-64.o -c rot13.c -march=rv32gc -mabi=ilp32d 
riscv32-unknown-elf-gcc  -g -Og -T spike.lds -nostartfiles -o rot13-64 rot13-64.o  -march=rv32gc -mabi=ilp32d
```

I am pretty sure that I have used the correct flags for RV32, as you can see in the `objdump` output for it:
```
riscv32-unknown-elf-objdump ./rot13-32 -f

./rot13-32:     file format elf32-littleriscv
architecture: riscv:rv32, flags 0x00000112:
EXEC_P, HAS_SYMS, D_PAGED
start address 0x10010000
```

timsifive, 2020-09-07:
> Connecting to the remote target in gdb fails with the error "fd requires xlen 4, but target has xlen 8".

This sounds like you're using an old gdb. I'd recommend trying mainline gdb from git://sourceware.org/git/binutils-gdb.git. I'm currently using change 85f783647061e, but later versions should work just as well.

mattwachter, 2020-09-08:
Hi, 
thank you for the quick reply.
This is how I compiled gdb from the repo:
```
mkdir build
cd build
../configure --program-prefix=riscv32-unknown-elf- --with-tui --target=riscv32-unknown-elf --prefix=$HOME/opt
make all
make install
```

Unfortunately, I still get the same error:
```
$HOME/opt/bin/riscv32-unknown-elf-gdb rot13-32 

GNU gdb (GDB) 10.0.50.20200908-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv32-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rot13-32...
(gdb) set arch riscv:rv32
The target architecture is set to "riscv:rv32".
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
bfd requires xlen 4, but target has xlen 8
(gdb) 
```
Here is the detailed debugging output for connecting to the remote target:
```
(gdb) set debug remote 1
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Ack
Packet received: PacketSize=4000;qXfer:memory-map:read-;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+
Packet qSupported (supported-packets) is supported
Sending packet: $vMustReplyEmpty#3a...Ack
Packet received: 
Sending packet: $QStartNoAckMode#b0...Ack
Packet received: OK
Sending packet: $Hg0#df...Packet received: OK
Sending packet: $qXfer:features:read:target.xml:0,1000#0c...Packet received: m<?xml version="1.0"?>\n<!DOCTYPE target SYSTEM "gdb-target.dtd">\n<target version="1.0">\n<feature name="org.gnu.gdb.riscv.cpu">\n<reg name="zero" bitsize="64" regnum="0" save-restore="yes" type="int" group="general"/>\n<reg name="ra" bitsize="64" regnum="1" save-restore="yes" type="int" group="general"/>\n<reg name="sp" bitsize="64" regnum="2" save-restore="yes" type="int" group="general"/>\n<reg name="gp" bitsize="64" regnum="3" save-restore="yes" type="int" group="general"/>\n<reg name="tp" bitsize="64" regnum=[3585 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:1000,1000#9d...Packet received: m"fa0" bitsize="64" regnum="43" save-restore="yes" type="FPU_FD" group="float"/>\n<reg name="fa1" bitsize="64" regnum="44" save-restore="yes" type="FPU_FD" group="float"/>\n<reg name="fa2" bitsize="64" regnum="45" save-restore="yes" type="FPU_FD" group="float"/>\n<reg name="fa3" bitsize="64" regnum="46" save-restore="yes" type="FPU_FD" group="float"/>\n<reg name="fa4" bitsize="64" regnum="47" save-restore="yes" type="FPU_FD" group="float"/>\n<reg name="fa5" bitsize="64" regnum="48" save-restore="yes" type="FPU_F[3585 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:2000,1000#9e...Packet received: mestore="no" type="int" group="csr"/>\n<reg name="sscratch" bitsize="64" regnum="385" save-restore="no" type="int" group="csr"/>\n<reg name="sepc" bitsize="64" regnum="386" save-restore="no" type="int" group="csr"/>\n<reg name="scause" bitsize="64" regnum="387" save-restore="no" type="int" group="csr"/>\n<reg name="stval" bitsize="64" regnum="388" save-restore="no" type="int" group="csr"/>\n<reg name="sip" bitsize="64" regnum="389" save-restore="no" type="int" group="csr"/>\n<reg name="snxti" bitsize="64" regnum=[3585 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:3000,1000#9f...Packet received: m"mhpmevent19" bitsize="64" regnum="884" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmevent20" bitsize="64" regnum="885" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmevent21" bitsize="64" regnum="886" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmevent22" bitsize="64" regnum="887" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmevent23" bitsize="64" regnum="888" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmevent24" bitsize="64" regnum="889" save-r[3585 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:4000,1000#a0...Packet received: m="hie" bitsize="64" regnum="1605" save-restore="no" type="int" group="csr"/>\n<reg name="htimedelta" bitsize="64" regnum="1606" save-restore="no" type="int" group="csr"/>\n<reg name="hcounteren" bitsize="64" regnum="1607" save-restore="no" type="int" group="csr"/>\n<reg name="hgeie" bitsize="64" regnum="1608" save-restore="no" type="int" group="csr"/>\n<reg name="htimedeltah" bitsize="64" regnum="1622" save-restore="no" type="int" group="csr"/>\n<reg name="htval" bitsize="64" regnum="1668" save-restore="no" typ[3585 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:5000,1000#a1...Packet received: l27" bitsize="64" regnum="2908" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmcounter28" bitsize="64" regnum="2909" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmcounter29" bitsize="64" regnum="2910" save-restore="no" type="int" group="csr"/>\n<reg name="mhpmcounter30" bits [... cut ...] ter31" bitsize="64" regnum="2912" save-restore="no" type="int" group="csr"/>\n<reg name="cycle" bitsize="64" regnum="3137" save-res[3574 bytes omitted]
bfd requires xlen 4, but target has xlen 8
(gdb) 
```

I suspect that the issue is not primarily with `gdb`, but that `spike` does not emulate a 32bit RISC-V hart, even when passed the `--isa=RV32IMAFDC` flag.
In the debug output shown above all registers are 64 bits long, for example the `zero` register:
`reg name="zero" bitsize="64" regnum="0"`

timsifive, 2020-09-08:
I regularly debug 32-bit spike through OpenOCD and gdb, so it is doable. I do this using a full 64-bit toolchain (compiler, gdb). My gdb says `GNU gdb (GDB) 10.0.50.20200106-git`. I'm using the latest spike, and the latest riscv-openocd.

If you attach your rot13-32 binary, I'll give it a quick try.

mattwachter, 2020-09-10:
That would be great, thank you.
[rot13-32.zip](https://github.com/riscv/riscv-isa-sim/files/5199785/rot13-32.zip)