riscv-software-src/riscv-isa-sim #545: setting mtval register when illegal instructions exception.

JaewonHur, 2020-09-15:
> On an
> illegal instruction trap, mtval may be written with the first XLEN or ILEN bits of the faulting
> instruction as described below

As the spec says writing `mtval` on illegal instruction exception is optional.
Just for notification, Spike does not write `mtval` when the exception while Rocket writes.

aswaterman, 2020-09-15:
I had been meaning to implement this feature but hadn't gotten around to it.  It's now here: a3376ff9af100f6f58ebdb18864eda6ac68b51af

cc @avpatel This also populates *tval on virtual-instruction traps.

JaewonHur, 2020-10-25:
I think a3376ff incorrectly sign extends instruction bits when illegal instruction exception thrown.

aswaterman, 2020-10-26:
@JaewonHur I believe that was fixed in ef3ea15aba0dc0a9f0fde85f1e21c8b98d76ace2

closed 2020-10-26 by aswaterman